fossasia/pslab-mini-hardware #7: Creating preliminary schematic

Tejasgarg, 2025-07-24:
This issue aims to make a schematic that has all the basic components for the schematic design in place 
This would include - 

- Main microprocessor (STM32H563RIT6)
- Power connections to the microprocessor along with decoupling capacitors(both the small capacitor values as well as a bulk capacitor for each array of power connections) 
- Ground connections
- Serial wire debug
- External Oscillator connections
- Reset switch control
- Boot switch control
- USB-C port connections, along with some ESD protection for it 
- Fuse 
- LED for power indication
- LED, which the microprocessor can control

Kindly suggest any points that I missed that should be added to the basic schematic

CloudyPadmal, 2025-07-25:
Voltage Regulator

Tejasgarg, 2025-07-25:
Yes, I plan to take reference from the old design to add some of the components and make the overall process faster