fossasia/pslab-mini-hardware #21: Oscilloscope input settling time

bessman, 2025-08-12:
The STM32H5 ADC has configurable sample period, from 2.5 ADC clock cycles to 640.5 ADC clock cycles. In order to achieve 5 MSps, we must use the fastest sample rate (2.5 ADC clock cycles) at the maximum ADC clock rate (75 MHz). In addition to the sample period, there is also a 12.5 conversion period. 75 MHz / (2.5 cycles/sample + 12.5 cycles/conversion) = 5 MSps.

The input signal must have settled by the end of the sample period in order for the measurement to be good. Can we simulate and verify that the settling time for a signal passing through the oscilloscope input circuit is < 2.5 cycles / 75 MHz == 33.3 ns?

Tejasgarg, 2025-09-01:
> I've looked into this a bit and I've come to the conclusion that the bandwidth of the current driver is way too low.
> 
> The TL082 has a GBW of 3 MHz, which gives a BW of 2.5 MHz in our configuration. The PGA further limits the bandwidth, especially at high gains. The combined bandwidth is about 2 MHz.
> 
> In order to capture a valid sample, the input signal must settle to 0.5 LSB within the ADC's track window. For 12-bits, this settling time is approximately 9 τ, or 716 ns for 2 MHz.
> 
> **That means the limiting component in the oscilloscope is not the ADC, it's the driver!**
> 
> With the current driver, we must increase the track window to at least 716 ns, which means the effective bandwidth of the oscilloscope becomes about 500 kHz.

Hi @bessman,
I was going through the calculations, and for achieving 0.5 LSB input settling within 33.3ns, I believe we would need something close to 45MHz.

Since following the formula τ = 1/(2pi*f) and then for 9τ to be under 33.3ns, I reached the value of 43MHz. 
Please let me know if I have overlooked anything or made a mistake in the calculations. Thank you!

Tejasgarg, 2025-09-01:
> [@Tejasgarg](https://github.com/Tejasgarg) any progress?

Hi @CloudyPadmal,

Yes, I have made selections for the replacement Op-Amps that I believe would fit our use case 
10MHz - 
ST TL084CPT

TI TLE2072CDR

45Mhz -
TI OPA828IDR

TI OPA828IDGNR

Kindly let me know your thoughts on this, thanks!

CloudyPadmal, 2025-09-01:
> Yes, I have made selections for the replacement Op-Amps that I believe would fit our use case 10MHz - ST TL084CPT

Isn't `ST TL084CPT` the same one we have right now (`TL082`) with 4 OpAmps instead of 2?

> 
> TI TLE2072CDR
> 
> 45Mhz - TI OPA828IDR
> 
> TI OPA828IDGNR

What is the difference between `OPA828IDR` and `OPA828IDRGNR`? Is it the same thing? Are they over $7 each?

CloudyPadmal, 2025-09-03:
> > > How about TI OPA356 or ADA4807-1? Both 200 MHz BW, $1-2 1ku cost.
> > 
> > 
> > These are much better.
> > [@Tejasgarg](https://github.com/Tejasgarg) could you please update the schematic with these?
> 
> Sure! I will update the schematic. These seem to be very good options. I have also been working on the input circuitry that will be required, along with these Op-Amps, to implement the +/-5V input range.
> 
> I believe we could implement something like a Summing Amplifier network for level shifting to reach the required ADC input pin ranges from the Op-Amp.
> 
> I have made a rough schematic for the same.
> 
> [image]

What does this circuit do? What will be the ADC input when Vin is +5 V and -5 V?

Tejasgarg, 2025-09-03:
Hi @CloudyPadmal,
Here is the schematic with proper resistor values

[image]

This circuit will convert 
-5v at Vin -> 0v at Vout 
and 
+5V at Vin -> 3.3v at Vout 

Kindly let me know if there are any shortcomings of this circuit that I have overlooked, thanks!

bessman, 2025-09-04:
What are VCC+ and VCC-? The ±6V rails used in the old hardware won't work for the OPA356, which has a rail-to-rail max of 5.5 V.

We can either just go with 3.3V / GND rails, with adequate margin in the output (might need to increase to 100 mV on either end), or supply it with something like -1V / 4V. In the latter case, we won't need much margin in the output.

Tejasgarg, 2025-09-05:
Ok, I will implement this with the new range.

New values for the resistor to achieve this range:
R18: 6.2k
R20: 20k
R40: 2M
R41: 620k

bessman, 2025-09-05:
That should be fine. Feel free to play around with the exact values to find good components, but keep input impedance in MOhm range and feedback resistance in 10k range.

bessman, 2025-09-05:
Here is an alternative which combines the divider output and adc input stages:

[image]

It's pretty hacky; it uses the Vref buffer negative rail as ground path in the DC case, which means we need one Vref buffer per channel. Still cheaper than the other alternative.

[simulation](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWK0CcAOFYAskUDYBmArBBAdgTxAUmpCwOoFMBaMMAKAHkQCAmWi3y88aEEKq1IHAIbgyVCSD5k+44SnBI2SMPHhRDe-RwKjeA9VX6DsWQ9IBOy1VYuDhtLGQ4Bjd2424lhi9nzQYChR0TExaKzsyHAQMMlQHADmASwh2fwOsuJgeB6KaKUgmmDa1eD6cA51JgBKRShqbCVtHZFqUvS0BNCM-TAIHM5s7UVdUz1gfSDlcBPdRQtrOWK0AvCZm7lz4vlSHADum+w9s1fpFyzlbg+CudL3j1szL9vnX8FinUEjyc4g+whY+SUtD0fnEfBw6w68O+9CS8BSsEgGLS0iyLGR-zhCIhizeRNsGyObA20n8JKekNy9lS+libNiaOxWOQnDxjIBkJO6QEZAsbjw9iU9gAao4mAAzDgAIzhaG0AjEaEYKWV4GwcMgVBIaj6HBF8kUtzsiNR0pkABd7RwAB7iQhFNBqPA1MiaMIygD2ABt7TIMkwADoAZ2aCpdFktaAgJFsZFCynsAEEACIAYWjACEAK7y+VMRyrNhYeyHa0LbYgMj1St1vgA60awwEZt0zuHTv11F4BJ9TFcjGce4DttFavgGdk-GQULtuevWFLsSDzfKZeosDQBAJcecu4GsSdqthPfSc1Kbe5Qcyh1O1o7-uUegNryjByH3X4mgVDwksJoEKK0iuiweCaBKIBwXgor+iAmaOkwAB2RYOgAlgG6G-Fe84AsIg7SGYAodOCkLWqc5FuNuwhwacd4dnuSh9iAspxhcKiUcBrifGSvGEmACiEmSomKI+YnPOk-iQsI-L7pi9SqckrAQWOp64mKnyCqSBEKdYfHpAGhhMQM6B0DAKTgKYYgEEgABitksEgNSccGobhhwQA)

bessman, 2025-09-05:
Minor improvement, connecting the bottom divider cap directly to ground: [simulation](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWc0BMYAcKAsBOAzDgGyQDskeWJeICkNIW1CApgLRhgBQA8iCuuhCsiIPHkJCRdSJwCG4LFkkTsdYRJzgk7JGHjwohvfs7jB-Qer4ChYRYZkAnPljUiLyw5U4Bja5fcbVixBJRRoMBwo6JiY9DYOKFhICBg4VM4Ac38hEJzWPBQHOVtiT1Z0Nw0tW119OAdwes4AJVscIvYy9g72oukGOjxoagGYBE5nHs6wbsiZsH6QSrhJvttF9eDBOhQTbOncy3mhQuKAdy2OGbnNmUuKqqFKo6hOB5ft0rU8++-XrpqF5OZ5PAqdKRNXxCVQbTqwr5KNLwVLJVHpN4HBF5VgIs7Sd4w1xwq53aHg8pnRFJeqxOmxGkpRmorgHKk4qn4t57EinCESWaeJQANUczAAZpwAEYw9C83GQTS4YoyzAw0iiGwZHkKJTsTbXEki2QAFxNnAAHpJNOwXigbWASLywiKAPYAGxNskyzAAOgBnFoSy0ubSOiTiB1UBguEAAQQAIgBhAMAIQAruLxcxHGt2PYcXYlIsdiAyCYpkXwPxbPY9qW8M0-ArQpZ69XQiBCAl+mjmUkuA92yXa8Wa38WwxjgWduThzXJydi9AsAl0UzkdqNVYR18R8aza11a2YQgJL8vGoYHt6rf0qXpNAJjLcegVEo1Sg8LyZFbWIRNEIJQgK7Z1YzjM1mAAO3TU0AEtXSgwl8zHAIBXHUxCEsKkRApFDijMTxdxEECCR1fD2ysdsRTFSVLgRdxiS+P5sUEKtmMJdi8nY4EsmPcBekndsCT-FAnS7bQUHtLtO02YU4KggMAAoAEaEGFABKThXUMUjBhwcwnwHIxTEECgQAAMVSIQkDDEBhQ9L0fU4IA)

bessman, 2025-09-06:
Grounding Cbot directly may not pole cancel right. Better to ground it via virtual ground: [simulation](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWc0BMYAcKAsBOAzDgGyQDskeWJeICkNIW1CApgLRhgBQA8iCiShCsiIPCjrDCUKJwCG4FOiEiU2ZVJzgk7JGHjxpdPfs55CS-oMl8BQsFiyHOAJz5rrl9dMqcAxjasVW1YsJUcUaDAcaJjY2PQ2DihYSAgYODTOAHMAoVDc1jEneXYSKXtHawqQTTBtOvB9OENGkwAlIXEldAkuhiU6Oiw6PGhqQeSEF07Ibt7ZhQGQHrhp1j6wRRmlKME6cXhs7fAcKz6+wc4Ad2PNpVLyrchrr2qq05kb96swMpO9l4Pf52P5gYgyVxVBx5HbQozPfzrBCPe4oZHAxzpeBpFI4jIyHJI8ofIl8BaXL5olG3J5+GHgaEhHYfTEpeBxDlxZL4rE4riE-K7ekXGTiEjgcrQwiOaqOABuWQAdgATTgAI066EcRWMfCwmQ16Ag62aeHQ4oGnDFizsmxqUnYdscADVZAAXN2cAAedkc4PQSkI1HCLoAls43dAADoAZwA4gA5AAi3r4CHuvwgeBImnY5gYepAAEEkwBhWMAIQArgAzGvMZxrN6-B1vER0MgmSFt1uMsHSPBNOkmjN-EdeP2JPa47mpZJcSkLazsRntwF9IEryotmSIjcicd5gtgaBYRJ4ue80WkLzL-LWF3uz0dQ870lAibDIQwA5NP+pJZBmgKYNXWI0ZggBBdABH1WBIbc-koEFg0LIsPWYRUq3dUMAHtFUBHtbzXMx7kFRl8nbGQSKIh0DzBCZnmtLc7Fo1spHlJVVRuVRKhUdx8mebj+J2LYmU+G0xLuIQehkWD+HFHRc1UDN0RlQRnVDRVYwACgARoQZ0AEp1yXftmKPBEJQZGViUEGU2X-JowDYcUgJ5I5+yFfsRUE+lqk8uFOBw6RpW8SAcAGSJkTRedpC4M1REcAAxNIhCQeoQGdHCABs3VkLJmE4IA)

bessman, 2025-09-06:
I think we can use TLV354x instead of OPA356 to save a little cost. Should be drop-in.

Tejasgarg, 2025-09-06:
> I think we can use TLV354 instead of OPA356 to save a little cost. Should be drop-in.

On LCSC, it seems the OPA356AIDBVR seems to be cheaper and more available.
Should we keep it the same in that case?
What do you suggest, @bessman?

bessman, 2025-09-06:
Whichever is cheaper.

bessman, 2025-09-07:
As decided during today's meeting, we'll do it the canonical way with buffered ladder: [simulation](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWc0BMYAcKAsBOAzDgGyQDskeWJeICkNIW1CApgLRhgBQA8g+na340sWEILoTOAQxB4UAoYxRihOcEnZIw8eFFjosCIyhxZS5EoTAk9kA0cI50CawhxkEN7Ts4AnECgkyqxEsvJioXTycH5hAqFyAmAiUAHwsYlicgzZrNlR6QDuOcGKufmcxZmshKLVFf6ZdJlY6KnRkJwA5nF8vbWpnXiEba0KdGNiYChtkgDGvRP5DMgs7MEwdobGhAjDODhghITUm5AQncVKqdd5ypeLvXdQsbeKQuI3JJwLk5+TRxWKGgYAOYPBEPQbA4eiOh20hBIWERJHQQVOsHOLx6AMIfDoA0kMk+rFa4DxpLaajAGhp4B0cFhjhMBF2rU8jmGTJwJgQuwMKAQWGwym86QASlNplMZlLgkcJDdmtBToNoAhYuxpes2lrgoSAjSYv49WIBqbFWhaJ0eha5WFBpV7TVRCTCU7AZTyY7ioDxj6VIqSQ1vV6PmS6FwFuxiGIyTHI3jRGc4BcQcycHyEAKhSLbFibVNY56FQ7JMUE2bXRx9cmfiBw20hIDk5j4BCOxDuRwCGRILRsOhEfmLt0G4nx2WXhXSy6iwII5x5DZG+AUl7RAA1XzMABmS9IT2SdXC7EsK23e84ACMpjy76Lz+hR7f5LrQco3KIhJ1lwEghEeLYPESYgJuUgAC4QZwAAecbyrs5IIcoojYGBAD2AA2EFSF0zAADoAM7inuzA7gAdnMzCwQwhAIW02YUkcSCoaIACCAAiADCREAEIAK67ruZGamSx5xm0KBoXQfbpCaZJSa6ZJ4JAyayAyY5eipSltE0LzRjWa7VtKikrGA0BQlqIAkLABBGHy8hEHgngkNCEiYmmXAziZaFeqZDymuJXriZ0Bk+cZyjaSsWCoNCEB4LAcJCgYJzJHg2iuew7mptO9pRVpqkvH+oTAeA96lVukHQZKgXrtYeIhUqgxqhqt7sPehmmc5NidHBJDUngknSskECsSAbFQcw5H8ZBACW6HkdI3qlSVZggNStJaAyyrnMyyIkGQiJCm5siJY4yXIuljAZSdYoxBWCloYC-kekBfLelFDyAlFCSFQ8v2iKtkjDE2C6g-awMjOtQFraE9WOnBgpIPISC0MjdhAqIJEAI78bNO6EUR5HMF0c0AG4EYRvhSLNmEAIQ0YK1A0gxT1WJjIBcQAFlI5HE5hhOEXMvNEYRPM7kRl77uhqQGmYTh6OlgqwqkXCDX0ABiFxiKjUwYdhuHUUAA)

bessman, 2025-09-08:
For protection, it would be ideal to clamp the op-amp inputs to the rails with schottsky diodes. But they'd need to be very low cap, single pF, and such components seem very expensive from what I can find. @Tejasgarg see if you can find reasonably priced low-cap schottskys.

Tejasgarg, 2025-09-08:
> For protection, it would be ideal to clamp the op-amp inputs to the rails with schottsky diodes. But they'd need to be very low cap, single pF, and such components seem very expensive from what I can find. @Tejasgarg see if you can find reasonably priced low-cap schottskys.

RB751S40 seems to be a viable option with low capacitance of range 2-2.5 pF and quite reasonably priced at $0.04

What do you think @bessman ?

bessman, 2025-09-09:
Come to think of it, the BAS70's high breakdown voltage would be quite nice to have. With a 10x, the input range would be ±50 V. If someone where to measure a voltage close to 50 V with a configurable 1x/10x probe, but accidentally set the gain to 1x, the RB751S40 would likely breakdown and fry the scope, while BAS70 would be OK.

bessman, 2025-09-09:
The attenuation block could look something like this:

[image]

Note also the 330 Ohm resistor before the ladder, which limits inrush current to the caps. It should be 100 - 330 Ohm.

The rails could be something other than -2.5 V / +5 V.

The 1k resistor between the tab point and the buffer limits clamp current over the diodes. We need to be careful not to introduce a low-frequency pole here. The total capacitance at the op-amp input node is around 6 pF (2 pF op-amp input + 2.5 pF for each diode). This creates a pole together with the ladder's Thevenin resistance, which at HF is equal to the resistor before the ladder. This pole must be >5x our target bandwidth, ideally x10.

With f_p == 50 MHz, R_s == 1 / (2pi * f_p * C) == 1 / (2pi * 50E6 * 7E-12) == 454 Ohm for the sum of the resistors before and after the ladder. Reducing the input resistor to 100 Ohm, the tap point resistor can be 330 Ohm. Does that limit the clamp current enough?

Assuming -50 V at the input, that's -15 V at the tap, +12.5 from the rail. I_clamp = 12.5 / 330 = 38 mA. Too high for RB751S40 (max 30 mA), OK for BAS70 (max 70 mA).

With 100 (before ladder) + 500 (after tab) Ohm instead, the pole is at 45 MHz. Should be OK. Clamp current 25 mA. OK for RB751S40, though uncomfortably close.

If we go with RB751S40 we need to be very careful here.

bessman, 2025-09-09:
The ADC buffer block could look something like this:

[image]

The OPA356 has an input capacitance of 1 pF, so there is a pole at 1 / (2pi * 1000 * 5E-12) ~= 30 MHz, which is a little low. We could reduce the resistor sizes a bit to bring it up. 1.2k resistors place the pole at 53 MHz.

The feedback cap needs to grow a little to account for the diodes' cap contribution at the input, to around 6 pF (caps + a little stray).

Tejasgarg, 2025-09-10:
Hi @bessman ,

I have been looking into values for resistors for the voltage divider that would work with a total of 1 M ohm and are also easily available in +-0.1% margin.

One of the feasible values is 820k ohm and 180k ohm divider.

Which would lead to +-0.9V out of the initial op amp.

Then replacing the input resistor with 0.9k ohm on the Vref input 
And 1.5k ohm on the Vatt input.
We can also keep the same values in the feedback loop as well.

This would keep the output in the range 0.15V at -5V at input 
And 
3.15 at +5V at input

bessman, 2025-09-10:
Looks good. Placing the tap lower also gives us more margin in overvoltage scenarios.

>  0.9k ohm on the Vref input and 1.5k ohm on the Vatt input.

Other way around, right?

We can increase the size of both the input and feedback resistors a bit (2-3x). I'm concerned the op-amp load might be a bit high. V_out / (R_s + R_g) = 3.15 / 2.4k = 1.3 mA. I'd like to keep it under 1 mA to make sure the output can swing to within 100 mV of the rails. Increasing the input resistance help lighten the load on the reference buffer.

The reason the input resistance had to be so low before was the clamp diodes, which added about 5 pF. But it turns out we don't need the clamps (see below), so we can up the resistors without getting a low frequency pole.

On closer reading of the OPA356 datasheet, **it has internal rail clamps**. So we don't need the schottskys after all. The internal clamps can handle up to 10 mA, to we need make sure we meet that.

With x0.18 attenuation, -50V -> 9V at the tap, 7V to rail. 7 / R_s < 10E-3 -> R_s > 700 Ohm. Let's make it around 2k. Without the extra diode capacitance, the pole is still well over 5 MHz. 1 / 2 / pi / 2E3 / 2E-12 ~= 40 MHz.

In the buffer stage, the resistors to Vatt and Vref are already sufficient.

For the reference buffer, we can use TLV9001 instead of OPA356. It's cheaper and better at driving capacitive loads.

Sorry there is so much back and forth in this circuit, but there are a lot of things to keep in mind and I want to make sure we get it right.

The current design with the above modifications ([simulation](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcsDMYBM6EA4GTClgOwBsJIeFIALChQKYC0YYAUAPI3XYibUUZe6fpCisAhr2w8+NOEP4BOECmh1RYePCjQSYbEQRFsGdAXR70OvQaMnhZnCWWatrAE69qo9JBnTePyheIjgPKRkg9ACwElFRajxIcOieFEh+b1F0kRok1gB3OR9hYpCrZKLUrx8A9CIKlICciJUECpCwgHNWlHbekh541hRBrhlS6m5wUmDkgGNW+qss8ppeJkFmVAwSdN8+hFoSJBg4CEqyltXlsSrmjKWGsU8byenZBKJWRamJzOmhASvGgYEU4IhkMh2E2EDAuiSvmwhAQg2cBh05zEPT+Kl84xUY2GkkY2FEzHIsRAFJAylU6nAWnk8JI1ChUPQhCIyFZ7MhnPqjLcACVqSwrIxFORmFsMENgt4VGo5joEOEZRKpWKtoRgo4wp4NdTdUbdT4wPkeqaJeKQKy5oVtRKTWTjY0ilSaVSzY7vRLXT6iqTsv74mJFhTyZ7YgkeCJYPA4bo2XzwQLuWd8NixXEaDxI3bcpcC4wTbbS41FvaafbpvGmanUzyU6n05iszA2D1azx7fbhkGY26c1HyMlfNywTbyMJR+sAGruegAM1YE9aQ+qcoXS9XACMxQEKVYLKJFKdWAewZOwPwEH0VNgx2vINzZ4FIp+F+IAC4-1gAB7gOgyjMCgb73mKKCZF4IAALIAJYACaMAAzgArgADphAA2ACegFlLed6lLe3L8KUACCAAiADCAA6qEAELocuy70O46q6kOFY1ME2C+PAXEnoqvFAsE+huFauriWJowST81JajxWrvhRsIgNg0BTPgoxSnskBmJAKCbGOCYXKCjqSjOom6u+xbKbmvFDgsSk2eSWriUoGmktp1DCIYfSciw+hEKZ7YXFZnn4mJ+Ljq+4CCFglK5sl35-qworMFM4BOWAswuQqYbFdAaoHswihwlS+XcqMkVASgLg5flVhkesVF-vQAB26G-ghAD2XUSHivCoolJ5jcoSCMOgpVClo7bnMI4K2PAjXGYodC+QgUrGUQRCxNEm3YNQ4WuEJRR2c8MmPJcN38LaLSXI9jy2mlySjDwU45uAigSv2YifeNo2UoI+Vji8w7vnZio+EJhqOVGzW5tk6SEcwJ3gNMYGUggEAPVY84IV1jEABQAI0IPOACUrA4blP3VlSGhMMomaGSBsRsukih+DgeAiIp25etEebrPCTKS1oYBMJOyaNn9XIRTobCLN9Xp-WLD2s82CucglmaRT027TNuAPJMbmum5r5uOkzlKizSz2O9VmtO4RmjGtysyljBgjse47gIT+9CMXu9DiEhrChWKyNI-w4kaNo52S-NEWsFaxF4vHKjyRo6P5cowVijtvDQeLVgACrzgAygXRBF4kOZvnWwEgP1nFAcwDe8HoJczrWbcAOLUZX6P7dSubY0YFcgJXAAy868-gZOQIxjCMfS86Me44gIThqG0w1FrgAJNCe-oSAEyA7AAAqUX0JCr+vm9qDThFIGYlKPGYyjX3fD9URkxmi-VCW9ab9WCADbwigZBzU7ApFAcZuQADELhiiTmKEA85+o4R-OILo9BWBAA)): 

[image]

Tejasgarg, 2025-09-10:
> We can increase the size of both the input and feedback resistors a bit (2-3x). I'm concerned the op-amp load might be a bit high. V_out / (R_s + R_g) = 3.15 / 2.4k = 1.3 mA. I'd like to keep it [...]

Here, instead of 4.5k and 2.7k ohms 
Can we instead use 3k and 1.8k ohms ?
Since, once again, 4.5k is not so readily available.

> Sorry there is so much back and forth in this circuit, but there are a lot of things to keep in mind and I want to make sure we get it right.

That is no problem at all.
Thank you so much for all the analysis and helping in developing the oscilloscope analog front end.

bessman, 2025-09-10:
The current draw from Vref is likely to be a considerable error source, so I'd like to make these resistors as large as possible while keeping the pole above 25 MHz.

R_th = R_att || R_vref = 1 / 2 / pi / f_p / C_in. With f_p = 25 MHz, C_in = 2 pF -> R_th ~= 3.2k

So aim for R_th = 1 / (1 / R_att + 1 / R_vref) as close as possible to 3.2k. We could go 10k / 6k. It's a little high, but a little over is better than a little under in this case.

PR #47 (2025-09-13): Updated Oscilloscope circuit
- fixes: #21 

This PR targets the updated input circuitry for the Oscilloscope 
Following the below circuit:

[image]

## Summary by Sourcery

Update the oscilloscope input circuitry in the KiCad project to match the redesigned circuit and include the required SMD3216-090N footprint module.

Bug Fixes:
- Resolve issue #21 by implementing the updated input stage for the oscilloscope.

Enhancements:
- Revise schematic and symbol files to reflect the new oscilloscope input circuit design
- Add SMD3216-090N footprint module to the PCB library for the updated component [... cut]